Hivemapper/odc-api #11: GNSS Auth: expose GNSS messaging sigs to ODC API

jayalberts, 2022-12-02:
Private keys on M9N to validate signature and hw is authenticity.

jayalberts, 2022-12-05:
solution from HB not good.
today auth message out of GPSD, flash keys work with John
Need to document process

jayalberts, 2022-12-07:
keys flashed onto dev camera. get dev build on cam. Slow but steady progress.

jayalberts, 2022-12-07:
Blocker was test device didn't have GNSS keys. No accurate documentation outside of hellbender. 

TN036 built before flashing keys. No UFL. No device ID EEProm.

jayalberts, 2022-12-08:
can install 3rd party modules to unpack GNSS. Today will be sampling.

BenDavidAaron, 2022-12-08:
Posting for others to gain context:
I have a script to sniff messages using `gpspipe -R`, so far I have not been able to sample any messages using `gpspipe` (which consumes them from `gpsd`) I'll be tweaking with the m9n configuration until I can get it to broadcast SEC-ECSIGN messages on the correct interface. So far I've tried UART1, sampling UART2 now.

BenDavidAaron, 2022-12-08:
Got the proper configuration commands from Caleb at Hellbender. Looks like this is ready to be folded into the ODC-API. Will open a PR soon. 
Docs on setting up ECSIGN Messages:
https://www.notion.so/How-do-I-enable-message-signing-on-HDC-d2520866b30842a3aa8b1f09031f1380

BenDavidAaron, 2022-12-09:
Have a PR ready for review (https://github.com/Hivemapper/odc-api/pull/8) that exposes binary GNSS messages to callers. 

This will also need some configuration changes to the M9N, I'm following up with Hellbender to figure out the best approach.

jayalberts, 2022-12-13:
Ready for firmware, need verification

BenDavidAaron, 2022-12-14:
Message contents verified, PR Merged 🚀

BenDavidAaron, 2022-12-14:
I don't know what release process this requires (since it's part of firmware). 
@punov can you weigh in?

jayalberts, 2022-12-15:
we already started testing on 0.11.11, so likely this included in next version for 0.11.12. Does this require any driving tests before it can be released widely?

jayalberts, 2022-12-16:
This has been merged

closed 2022-12-19 by BenDavidAaron