Hivemapper/odc-api #92: Triage HDC memory leaks

punov, 2023-09-20:
- Happens after ~20min
- Investigate yappi logs
- Try valgrind for identifying memory issues
- Add loaded images and model stats/snapshots for ml stuff

punov, 2023-09-20:
When looked closer to the memory usage, 
on every iteration over the folder, memory is getting filled with the imagery that is eventually getting cleared, but not before the next iteration.

Having an extra swap space helps to mitigate that, all the memory is eventually getting freed up, + some additional garbage collector executions

[image]

jayalberts, 2023-09-21:
no new tickets to create, nothing to release.

closed 2023-09-21 by jayalberts