Hivemapper/odc-api #102: PoC use SQLite instead of log files

jayalberts, 2023-10-09:


jayalberts, 2023-10-12:
Going well, easy to install/code/fetch/refactor. Almost done. 

Everything merging into a single table. Can we quickly make 2 separate tables for sensor data.

jayalberts, 2023-10-13:
was able to separate tables, need to finalize and create PRs today

jayalberts, 2023-10-16:
haven't been able to work on it, Charles and Eduard answered and so now unblocked.

jayalberts, 2023-10-17:
Aiming to wrap it up today.

jayalberts, 2023-10-19:
not a chance to finish yet

punov, 2023-10-20:
Data-Logger v1.2.0 with separate tables for sensor data:
https://github.com/Hivemapper/hivemapper-data-logger/pull/2

ODC API v3.7.0 to support SQLite:
https://github.com/Hivemapper/odc-api/pull/112

jayalberts, 2023-10-20:
ready. to be discussed with Evan in person.

closed 2023-10-23 by jayalberts