Hivemapper/odc-api #124: MotionModel Clean-up

punov, 2023-11-13:
We're revisited with Evan the code of the motion model,
proposed approach is here: 
https://gist.github.com/evanmoss/2a1a2f8d7de85fef5796e7b4371a3e6a

Solution is going to be cleaned up according to approach proposed.

jayalberts, 2023-11-13:
let us know about testing requirements

jayalberts, 2023-11-14:
discovering defects, fixing, drive testing this afternoon.

jayalberts, 2023-11-16:
issues with continuous framekm: 10s period of sensor info missing related to SQLite migration / data logger.

need to pre filter data before inserting into table to mitigate problem, currently in testing.

jayalberts, 2023-11-17:
still having unexpected symbols in ublox. fixed side effects but not root cause.

punov, 2023-11-17:
Solution is ready to be wrapped for this round of testing:
i will share the latest code with Evan once cleaned up.

Unexpected UBLOX symbols will be addressed with the next iteration

jayalberts, 2023-11-27:
checking out logs
main problem is camera trying to pack recent framekm but it fails because no images are ready, no proper handling to move on so it gets stuck trying to pack same framekm over and over. 
Fix is straight forward.

S is doing better on continuous numbers. HDC chunking more often because less frames to select from, especially if FPS or GPS drops.

closed 2023-12-06 by jayalberts