Hivemapper/odc-api #127: Method to call gyro calibration from odc-api

jayalberts, 2023-11-21:


evanmoss, 2023-11-21:
- In the naive case, we can assume the vehicle isn't moving when the power first comes on and call the calibration for gyro on boot up.

- A better solution is to opportunistically detect no movement and periodically calibrate the IMU when it hasn't been calibrated yet (or hasn't been calibrated in _t_ minutes (30?).

jayalberts, 2023-11-28:
Added ODC service to calibrate gyro and accel. Seems to be working, end of today should have PR.

jayalberts, 2023-12-01:
required permissions, running into build issues.

jayalberts, 2023-12-04:
was able to build new image, was able to test and it works. submitted PR.

jayalberts, 2023-12-05:
PRs merged

jayalberts, 2023-12-08:
Masaya has a test written for this

jayalberts, 2023-12-15:
Released in HDC 3.8.1, still need to include for HDC-S release

jayalberts, 2024-01-08:
released in 3.8.17 for HDC-S

closed 2024-01-08 by jayalberts