Hivemapper/odc-api #131: GNSS Auth - ODC

punov, 2023-12-02:


jayalberts, 2024-02-02:
@punov to share more info

Masaya-RT, 2024-02-08:
Some initial deep-dives into data-logger
Don't see any GNSS secure documentation. Evan knows where NDA-locked doc version is. Will get Mazar access.

Message is emitted that can be grabbed & validated

jayalberts, 2024-02-09:
documentation from Evan is good. Question about where to get public key -- ask Evan.

jayalberts, 2024-02-12:
Got key from HDC camera - moving forward

jayalberts, 2024-02-13:
Module sends sign messages and able to parse. 

Next: writing to verify signature.

Per framekm send 1 signed message and public key with framekm metadata.

When processing on backend, look up key to verify signature. 

Create new tasks for task tracking.

jayalberts, 2024-02-15:
Got hashing working as part of verification.