Hivemapper/odc-api #177: move edge privacy detections to uploaded fkm payload

Masaya-RT, 2024-03-14:
Sending all privacy via metadata

Move to concat in FrameKm -> add logic to pull that out from there.

Should save DB size/fees

evanmoss, 2024-03-14:
If it's added at the end, this routine should still be safe as is, and probably just need to add a new method that does the same thing but starts the next index (i.e., grab the rest of the bytes)

https://github.com/Hivemapper/network/blob/main/server/src/models/FrameKM.ts#L459

https://github.com/Hivemapper/network/blob/main/server/src/models/FrameKM.ts#L55C10-L55C28

Masaya-RT, 2024-04-22:
Mazar trying to start today.
Actually already done by Alexei; should add to exif data now.

Should now make changes in backend

closed 2024-04-22 by Masaya-RT