Hivemapper/odc-api #200: Add "metadata" for sensor telemetry packets

mazar-farran, 2024-05-16:


mazar-farran, 2024-05-16:
This is actually more complex than I thought. S3MultipartUploader() takes a file path so I can't arbitrarily change the format of the JSON.

We need to create separate files for metadata and sensordata

Masaya-RT, 2024-05-17:
Idea: Single, flat array of all fields in SensTel
- This will factor out fields common to all entries

Has a draft PR, but found issues after. Revising more to fix

Masaya-RT, 2024-05-20:
The PR is ready for Review. After approval, do E2E testing to make sure this is working.

Masaya-RT, 2024-05-21:
Has been reviewed, rebased in app 7.5.

Wants to take look at old firmware compatibility

Masaya-RT, 2024-05-22:
Merged to 7.5
Next step to give to CTP

Mazar's best approach rec: 
- Ensure no changes/incompatibility: Make sure all changes are deployed to CTP so that next FW doesn't cause issues

Masaya-RT, 2024-05-23:
Waiting for CTP release, moved to ready cc @HariHive

mazar-farran, 2024-07-11:
Done

closed 2024-07-11 by mazar-farran