Hivemapper/odc-api #202: Use Sensor Fusion approach to detect stationary and filter out at motion model (i.e., needs to be moving to add to the model)

Masaya-RT, 2024-05-20:
We want to filter out data where we have "dancing position" when car has GPS but is stationary

Masaya-RT, 2024-05-20:
Kalman filter: Filter wasn't stopping the issue, but was reducing the error radius.
- Problem is and has existed, but filter helps

Masaya-RT, 2024-05-21:
Not many updates
Question for Alexei: Start of the drives, you query sensor logs and see motion model. Is it filtered out there? Yes

Masaya-RT, 2024-05-22:
Inital experimentation is good. Python is proven

Solid portion converted to TypeScript, but Python Math still needs migration. Trying to figure out low-pass filter implementation.

Needs to be widely-applicable and relatively noise-proof.

Masaya-RT, 2024-05-23:
Everything at this point is written and tested in TypeScript. 
Performance is slower than Python
Roger to discuss with Alexei to weave into motion controller (tradeoff implications)
Should be ready for testing after

Masaya-RT, 2024-05-28:
Still waiting on feedback from the branch

roger-berman, 2024-05-31:
Got Data from @punov  and will be familiarizing myself with @mazar-farran's new test framework to begin debugging performance

Masaya-RT, 2024-06-03:
Results still pending - Focusing on Data Logger refactor

Masaya-RT, 2024-06-04:
Ran E2E test, was breaking.
- To talk to Mazar to figure out issue